OllisGit/OctoPrint-SpoolManager #335: Unable to save spool after copying from template

markusbuschhoff, 2023-07-25:
v1.70, adding spools from scratch works. However, marking a spool as template and then adding/copy template and modifying the spool does not save it. I modified display name, color, switched to "Filament Amount" calculations, entered start/end weight (with empty spool weight filled from template). All fields get calculated correctly. Saving fails (spool not showing up). Log excerpt:

2023-07-25 15:58:23,703 - octoprint.plugins.SpoolManager - INFO - API Save spool
2023-07-25 15:58:23,707 - octoprint.plugins.SpoolManager - INFO - Load spool for update with database id ''
2023-07-25 15:58:23,727 - octoprint.plugins.SpoolManager - WARNING - Save spool failed. Inital loading not possible, maybe already deleted.
2023-07-25 15:58:23,728 - octoprint - ERROR - Exception on /plugin/SpoolManager/saveSpool [PUT]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/api/SpoolManagerAPI.py", line 973, in saveSpool
    newDatabaseId = self._databaseManager.saveSpool(spoolModel, withReusedConnection=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 1068, in saveSpool
    totalWeight = spoolModel.totalWeight
AttributeError: 'NoneType' object has no attribute 'totalWeight'

markusdd, 2023-08-07:
I have the exact same prolem and it is annoying as f* xD

scaytrase, 2023-09-06:
Same here saving spool from template doesn't work

markusdd, 2023-10-07:
This is still an issue.

eazrael, 2023-11-14:
Same issue, workaround by markusdd works. 

Project seems quite dead

broderickhyman, 2023-12-11:
This plugin is under new management at [dojohnso OctoPrint-SpoolManager](https://github.com/dojohnso/OctoPrint-SpoolManager) where I have fixed the template issue in the [1.7.3 release](https://github.com/dojohnso/OctoPrint-SpoolManager/releases/tag/1.7.3). The new management is now registered in the [official Octoprint plugins list](https://plugins.octoprint.org/plugins/SpoolManager/) so you can install like normal.

closed 2024-01-21 by github-actions[bot]