OllisGit/OctoPrint-SpoolManager #337: Failure during installation -> ERROR: Failed building wheel for OctoPrint-SpoolManager

FreunderNacht, 2023-08-08:
**### What is the problem?**

When I try to install the spool manager, the installation is aborted.

The following is written in the log of the plugin_pluginmanager_console (Extract from the log file):

```
2023-08-07 07:50:40,950 ! The above exception was the direct cause of the following exception:
2023-08-07 07:50:40,950 ! 
2023-08-07 07:50:40,950 ! Traceback (most recent call last):
2023-08-07 07:50:40,950 ! File "<string>", line 2, in <module>
2023-08-07 07:50:40,950 ! File "<pip-setuptools-caller>", line 34, in <module>
2023-08-07 07:50:40,950 ! File "/tmp/pip-req-build-tl1f8prq/setup.py", line 103, in <module>
2023-08-07 07:50:40,950 ! setup(**setup_parameters)
2023-08-07 07:50:40,950 ! File "/home/ssh_octoprint-1/OctoPrint/lib/python3.10/site-packages/setuptools/__init__.py", line 153, in setup
2023-08-07 07:50:40,950 ! return distutils.core.setup(**attrs)
2023-08-07 07:50:40,950 ! File "/usr/lib/python3.10/distutils/core.py", line 148, in setup
2023-08-07 07:50:40,950 ! dist.run_commands()
2023-08-07 07:50:40,950 ! File "/usr/lib/python3.10/distutils/dist.py", line 966, in run_commands
2023-08-07 07:50:40,950 ! self.run_command(cmd)
2023-08-07 07:50:40,950 ! File "/usr/lib/python3.10/distutils/dist.py", line 985, in run_command
2023-08-07 07:50:40,950 ! cmd_obj.run()
2023-08-07 07:50:40,950 ! File "/home/ssh_octoprint-1/OctoPrint/lib/python3.10/site-packages/wheel/bdist_wheel.py", line 401, in run
2023-08-07 07:50:40,950 ! self.egg2dist(self.egginfo_dir, distinfo_dir)
2023-08-07 07:50:40,950 ! File "/home/ssh_octoprint-1/OctoPrint/lib/python3.10/site-packages/wheel/bdist_wheel.py", line 544, in egg2dist
2023-08-07 07:50:40,950 ! pkg_info = pkginfo_to_metadata(egginfo_path, pkginfo_path)
2023-08-07 07:50:40,950 ! File "/home/ssh_octoprint-1/OctoPrint/lib/python3.10/site-packages/wheel/metadata.py", line 160, in pkginfo_to_metadata
2023-08-07 07:50:40,950 ! for key, value in generate_requirements({extra: reqs}):
2023-08-07 07:50:40,950 ! File "/home/ssh_octoprint-1/OctoPrint/lib/python3.10/site-packages/wheel/metadata.py", line 138, in generate_requirements
2023-08-07 07:50:40,951 ! for new_req in convert_requirements(depends):
2023-08-07 07:50:40,951 ! File "/home/ssh_octoprint-1/OctoPrint/lib/python3.10/site-packages/wheel/metadata.py", line 103, in convert_requirements
2023-08-07 07:50:40,951 ! parsed_requirement = Requirement(req)
2023-08-07 07:50:40,951 ! File "/home/ssh_octoprint-1/OctoPrint/lib/python3.10/site-packages/wheel/vendored/packaging/requirements.py", line 37, in __init__
2023-08-07 07:50:40,951 ! raise InvalidRequirement(str(e)) from e
2023-08-07 07:50:40,951 ! wheel.vendored.packaging.requirements.InvalidRequirement: Expected end or semicolon (after version specifier)
2023-08-07 07:50:40,951 ! pillow>=6.2.0<7.0.0
2023-08-07 07:50:40,951 ! ~~~~~~~^
2023-08-07 07:50:40,951 ! [end of output]
2023-08-07 07:50:40,951 ! 
2023-08-07 07:50:40,951 ! note: This error originates from a subprocess, and is likely not a problem with pip.
2023-08-07 07:50:40,951 ! ERROR: Failed building wheel for OctoPrint-SpoolManager
2023-08-07 07:50:40,951 ! ERROR: Could not build wheels for OctoPrint-SpoolManager, which is required to install pyproject.toml-based projects
```

Here is the full log-file:

[plugin_pluginmanager_console.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/12290549/plugin_pluginmanager_console.log)

The other plug-ins could be installed without problems (e.g. PrintJobHistory, DisplayLayerProgress Plugin, Cost Estimation, Printoid Plugin, Slicer Thumbnails, etc).

The only one that causes problems is the spool manager.
It is the only one that cannot be installed.

**### What did you already try to solve it?**

I have restarted Octoprint several times.
I have restarted the whole machine several times.

To make sure that there are no leftovers,
I tried uninstalling the plugin:

[Forum -> I installed a broken plugin, how do I uninstall it again?](https://community.octoprint.org/t/i-installed-a-broken-plugin-how-do-i-uninstall-it-again/217)

Here I received the following output:

```

ssh_octoprint-1@Octoprint-1:~/OctoPrint/bin$ pip uninstall OctoPrint-SpoolManager
WARNING: Skipping OctoPrint-SpoolManager as it is not installed.
ssh_octoprint-1@Octoprint-1:~/OctoPrint/bin$ pip uninstall OctoPrint-SpoolManager
WARNING: Skipping OctoPrint-SpoolManager as it is not installed.
ssh_octoprint-1@Octoprint-1:~/OctoPrint/bin$ pip uninstall octoprint_SpoolManager.api
WARNING: Skipping octoprint_SpoolManager.api as it is not installed.
ssh_octoprint-1@Octoprint-1:~/OctoPrint/bin$ pip uninstall octoprint_SpoolManager.models
WARNING: Skipping octoprint_SpoolManager.models as it is not installed.
ssh_octoprint-1@Octoprint-1:~/OctoPrint/bin$ pip uninstall octoprint_SpoolManager
WARNING: Skipping octoprint_SpoolManager as it is not installed.
ssh_octoprint-1@Octoprint-1:~/OctoPrint/bin$ pip uninstall octoprint_SpoolManager.common
WARNING: Skipping octoprint_SpoolManager.common as it is not installed.
```

**### Have you tried running in safe mode?**
No

**### Did running in safe mode solve the problem?**
No

**### Systeminfo Bundle**

[octoprint-systeminfo-20230807203755.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/12290553/octoprint-systeminfo-20230807203755.zip)

**### Additional information about your setup**
<small>OctoPrint version, OctoPi version, printer, firmware, browser, operating system, ... **as much data as possible**</small>

Octoprint runs in an unprivileged LXC container on Proxmox.

Operating system: Ubuntu 22.04 LTS

OctoPrint 1.9.2

Python 3.10.12

Printer: Anycubic Mega S

Octoprint was installed with the help of the following script: [octoprint_deploy](https://github.com/paukstelis/octoprint_deploy)

**###**

Can anyone give me a tip as to why I can't get the spool manager installed and how I can solve the problem?

I have already tried to study the log, but unfortunately I can't find anything there.

I have also searched the internet, but I couldn't find anything.

Thanks

Freund der Nacht

FreunderNacht, 2023-08-08:
See the issues: #331 #327 #324 

See also:
[OctoPrint Forum -> Spoolmanager Plugin -> Failure during installation](https://community.octoprint.org/t/spoolmanager-plugin-failure-during-installation/53307/6?u=freunddernacht)

mdziekon, 2023-08-08:
The issue has been fixed already in my fork of this plugin, see: https://github.com/mdziekon/OctoPrint-SpoolManager

Impact123, 2023-09-08:
Unstale.

closed 2023-10-19 by github-actions[bot]