OllisGit/OctoPrint-SpoolManager #346: Fails build through plugin manager or URL

alien8r, 2023-12-02:
Building wheel for OctoPrint-SpoolManager (setup.py): finished with status 'error'
Running setup.py clean for OctoPrint-SpoolManager
error: subprocess-exited-with-error

× python setup.py bdist_wheel did not run successfully.
│ exit code: 1
╰─> [215 lines of output]
Found packages: {'octoprint_SpoolManager.common', 'octoprint_SpoolManager', 'octoprint_SpoolManager.api', 'octoprint_SpoolManager.models'}
running bdist_wheel
running build
.....
File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/wheel/vendored/packaging/_tokenizer.py", line 163, in raise_syntax_error
raise ParserSyntaxError(
wheel.vendored.packaging._tokenizer.ParserSyntaxError: Expected end or semicolon (after version specifier)
pillow>=6.2.0<7.0.0
~~~~~~~^

alien8r, 2024-01-02:
Tried various fixes, all useless.
No input at all from maintainer.

drohhyn, 2024-01-08:
There is a fork which is already in the plugin repo: https://github.com/dojohnso/OctoPrint-SpoolManager

closed 2024-02-18 by github-actions[bot]