OllisGit/OctoPrint-SpoolManager #354: [Feature Request] Don't spam "SPM:WARNING: Filament not enough!" warnings

NewsGuyTor, 2024-08-28:
Every time I open Octoprint to check print process on a print, started on minus measured filament (but still plenty to complete the print), I get a warning and notification. 

I don't see the point of that, considering I consciously disregarded all the warnings of that exact thing, at the start of the print.

WildRikku, 2024-09-11:
The current version of SpoolManager has an option to deactivate those warnings. If you don't have it, you are using an outdated version. You can get the current one through OctoPrint's plugin manager.

NewsGuyTor, 2024-09-24:
No. I don't want to hide the warning popup and notification before starting a print, I want to hide the notification that appears every single time a new Octoprint tab is opened during print.

WildRikku, 2024-09-28:
I see. This plugin has a new home at https://github.com/WildRikku/OctoPrint-SpoolManager/, please open a bug report there, linking this issue here. Please name your OctoPrint and SpoolManager version and your browser. Please close this issue afterwards as this repository is no longer maintained.

NewsGuyTor, 2024-09-28:
Done! :) https://github.com/WildRikku/OctoPrint-SpoolManager/issues/17

closed 2024-09-28 by NewsGuyTor